makerbase-mks/MKS-DLC #1: Missing documentation

redwil, 2019-04-12:
If this is open source than please provide the sources.

I have searched the internet to find out more information about MKS-DLC but without big success.
This  repo contains only windows driver software, hex files and some old grbl software for uploading to arduino but nothing about your MKS-DLC v2.0 board

Mostly what you find are seller of some MKS variants and some conflicting pinouts for connecting to various CNC/3Dprinter hardware all in Chinese with some automatic English translations which, btw, are quite poor.

Please provide
1) Complete schematics and especially pinouts at connectors.
2) connector pins to MCU port pins relation
3) power max/min ratings
4) software sources (that can be build without guessing that means i.e a complete Makefile)
5) we do not need fabrication documents (i.e. gerbers and all)

BTW we are not interested in building the hardware or even to get inspired from it we need that info to be able to use it.

thorflea, 2019-05-08:
> If this is open source than please provide the sources.
>
> I have searched the internet to find out more information about MKS-DLC but without big success.
> This repo contains only windows driver [...]

Yes, it has been frustrating finding anything useful on the internet.  It is so bad that I am looking at getting a tinyG controller and replace the MKS DLC v2.0.

colinluthier, 2019-05-11:
[image]
 This is all I can find...

thorflea, 2019-05-12:
There are several posts on FB SMW3D Hobby-Fab CNC group with more details.
About this controller.

redwil, 2019-05-23:
Thank you guys. 

@colinluthier I found that too but is a rev 1.0 not 2.0 which does not have the on/off button.
@thorflea could you post a link to that group?

The problem remains with respect of the 'Open' part of the board. No sources and no schematics or pinout, besides the confusing/mismatching information.

We are way too lenient vs various unfair/shady practices of companies/peoples who take advantage of the real open source/hardware without even providing the necessary information to properly use the product, let alone take full advantage of it.

makerbase-mks, 2019-10-22:
We will open the MKS DLC hardware, and then.

redwil, 2019-11-21:
@makerbase-mks After 7 (seven) months an ambiguous comment: 
'.... , and then': then what? not even a slight comment on the 'openness'. 

This is unbelievable rude and shows a total lack of respect against the ones that contributed/donated their time/brains/resources towards bettering the world.

Better declare it as proprietary and be done with it. At least people will know to stay away from your products.

makerbase-mks, 2019-11-22:
This may be caused by incorrect translation. DLC Hardware Reference: https://github.com/makerbase-mks/MKS-DLC/tree/master/hardware
MKS DLC is directly using GRBL

closed 2019-12-03 by makerbase-mks

hipop, 2021-07-23:
I find this：https://raw.githubusercontent.com/microrobotics/MKS-GBRL/master/MKS-GBRL-Board-Layout-V2.pdf