makerbase-mks/MKS-DLC #6: Dual Y inverted way

UnleashSpirit, 2020-12-12:
Hi, I jsut switch from CNC shield to MKS DLC 2.0.
Flash last firmware, MKS_TFT35_CNC_V1.0.2 and last grbl 1.1h
My Y1 and Y2 are turning oposite way
Is there a way to invert only Y2 (without switching cables) ?

DirtyEngineer, 2020-12-12:
Yes - Use the $3 setting in grbl to invert a specific axis.

mks-viva, 2020-12-14:
@UnleashSpirit 
Hi! I am sorry, because the hardware has determined that Y1 and Y2 always move in opposite directions.
[image]
[image]

mks-viva, 2020-12-14:
So, you need change the cable.

closed 2020-12-14 by mks-viva