makerbase-mks/MKS-DLC #7: dlc v2

llocken45, 2020-12-17:
G28 is reversed from end stops $H homes correctly however TFT35 uses G28 how should i fix without rewiring?

mks-viva, 2020-12-18:
Hi! You need update firmware for your MKS TFT35 dspialy screen : https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS-TFT-laser

llocken45, 2020-12-18:
my firmware is updated
After I read through the doc https://lightburnsoftware.github.io/NewDocs/CommonGrblSetups.html 8 I found that I can just use “G10 L2 P1 …” command to shift the X/Y to positive and to the dimension of my machine (somehow its different form my $130-$132 position), and it works, since it shifted the view point of the GRBL’s CNC negative coordination system to positive.

So just do
G10 L2 P1 X[at Home] Y[X at home]

Like my machine I measured them as X=343 and Y=470 (my far up right), I will enter the following:
G10 L2 P1 X-343 Y-470

in light burn use get position after running $H to edit x and y

closed 2020-12-18 by llocken45