makerbase-mks/MKS-DLC #9: Wiring limit switches of the MKS - DLC

mldmld68, 2021-01-03:
Hi

Where could I find a documentation about the limit switches of the MKS-DLC ?
Do the configuration to use is the same as : https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches ?

In particular, do one need to add filtering with the MKS - DLC ? Or  do the MKS-DLC already have filtering features ? 

AFAIU, one need to enter $5=1 to use Normaly Closed switches.
Thank you

mks-viva, 2021-01-04:
Hi! MKS DLC compatible with GRBL

UnleashSpirit, 2021-01-04:
It works for me, I use basic limit switches (like 3D printer) wired in a NO way. On the board wire them in the two lower pins, G&S, of the 3 pins connectors
My $5 is at 0.

mldmld68, 2021-01-04:
Thank you,  so no need of capacitors and resistors.
I would consider NC option instead.

mks-viva, 2021-01-05:
@UnleashSpirit 
Thank you!

@mldmld68
It is usually triggered by S G. If you actually test it and it works, then you can use.

closed 2021-02-21 by mks-viva