makerbase-mks/MKS-DLC #14: laser 3 pins

danyal-hmeedan, 2021-05-05:
how to wire a 3 pins laser .

mks-viva, 2021-05-08:
12V GND TTL？
If yes, you need connect to board's 12V GND and TTL signal
[image]

albertolg, 2021-06-25:
what is the Laser output? I connected my laser module there and after a couple of seconds it burned

Igoravg98, 2021-07-12:
You will need to solder one more wire to the laser's GND and connect one of these wires to the TTL output along with the PWM. You will end up with 2 wires for the laser and 2 for the TTL output
I was having the same problem, I did it and solved it.

AussieMakerGeek, 2021-10-29:
> You will need to solder one more wire to the laser's GND and connect one of these wires to the TTL output along with the PWM. You will end up with 2 wires for the laser and 2 for the TTL output I was having the same problem, I did it and solved it.

Can you elaborate? I don't understand what you mean

8tabbie, 2023-03-29:
#You will need to solder one more wire to the laser's GND and connect one of these wires to the TTL output along with the PWM. You will end up with 2 wires for the laser and 2 for the TTL output I was having the same problem, I did it and solved it.

I also need more info please. I damaged my Makerbase MKS DLC32 V2.1 which is 3 pin, with a Makerbase MKS DLC V2.1 which is 4 pin (could not find DLC32 in SA). Now I cannot read/connect to my laser. I have tried GS and +12V0-, no signal. I have tried -12V0- and S, no signal. Extra VGS, get signal, but can't connect' (assuming because 5V). Please is someone can assist.

AussieMakerGeek, 2023-03-30:
You use the -12Vo+ and TTL connections. I'm not sure what you mean by the DLC 2.1 being 4 pin (where?) I think you mean 3 pin and 2 pin.

I have both the DLC32 and DLC v2.1

DLC32 has TTL-G-12v connection for Laser
[image]

DLC 2.1 has -12Vo+ and the TTL on the 'S' pin of the TTL connector.
[image]

You can use the 'Laser' connector instead, that is just TTL and 12v and you need a ground from another connector but I think the above is easier.

8tabbie, 2023-03-30:
Good morning.
Thank you for your reply.
I have tries you first option, but not getting a signal.
Which one on the laser is TTL (+ or -)?

Yes, I meant DLC 2.1 2x2 pin.

DLC 2.1 battling to get signal and read laser.

DLC32 2.1 it does not read the board - USB port not reading?

Thank you for your assistance.

Hoping to hear from you soon.

Regards

Louis

AussieMakerGeek, 2023-03-30:
The - on the laser connection is not actually TTL sorry, it's an open collector output which produces a negative PWM signal to ground. An exact opposite of the TTL output which is a 5v PWM.

If you leave the laser disconnected and then test fire it with software, the LED next to the FET, labelled D1 should light up too.  If it's not doing that, you're not going to get the laser to fire.

If you've just got a basic diode laser - Apply the 12v and ground, then just blip the TTL wire on the laser to a 5v pin and see if it fires. You might be struggling with it because the laser itself is bad. MAKE SURE YOU ARE FOLLOWING SAFETY GUIDELINES FOR WORKING WITH A LASER BEFORE DOING THIS.

PS - I would recommend editing your reply to remove the personal details.

8tabbie, 2023-03-30:
I am using the Lasertree LT-4LDS-V2 Laser.

8tabbie, 2023-03-30:
[image]

Is this the light that must come on when testing?

8tabbie, 2023-03-30:
[image]
When I connect to USB D4 red LED comes on and D5 blue LED flashes. When I ping/connect from LaserGRBL (not connected) the D7 blue LED flashes. Nothing on D1

8tabbie, 2023-03-30:
R12 R13 R14 R15 and R26 give me a signal on the Driver Adaptor P-DA-01, but still not reading the laser.

8tabbie, 2023-03-30:
Other problem. Makerbase MKS DLC32 V2.1 - Does not read the USP port. I think I need to run Firmwear, but I am not sure how to do this? With DLC my system shows the USB port CH340 Driver, but not with DLC32?

8tabbie, 2023-03-30:
[image]
I get a signal from this yellow wire. Machine still not connecting to laser.

8tabbie, 2023-03-30:
[sessionlog.txt](https://github.com/makerbase-mks/MKS-DLC/files/11110142/sessionlog.txt)
This is my session log. See date 21/03/22. Problem started around 12:00. I got back at around 4. Since then I can't get it to work.

8tabbie, 2023-03-30:
Can my problem not be firmware? I have no idea how to do that? Common problem for me is that the signal is not getting to the board or USB port?

ATR7000, 2023-05-20:
Hi. I am having the same problem.  
Trying to connect a 3pin laser to MKS DLC 2.1. 
Did you figure out how to finally connect it?
Cheers....

8tabbie, 2023-05-21:
Hi, I bought a new Makerbase MKS DCL32 V2.1 and my problem were solved. I now keep 2 boards in stock just in case. Could not get the Makerbase MKS DLC 2.1 to work.