makerbase-mks/MKS-DLC #23: Z probe

SlavaKot, 2022-01-25:
Hello everyone. which pin is responsible for the Z probe on the MKS-DLC board.

raindancer2204, 2022-01-30:
Hello everyone, my question goes in the same direction. I have a Makerbase MKS DLC **2.1** with your CNC OS MKS_TFT24_CNC_V1.0.3.2 .
I would now like to take advantage of the Z Probe.

It says sample at the bottom right, I've marked it in red. Where do I have to connect the 2 cables from the probe? It says V G S what goes where? I thank you in advance.
Here is a picture of the mainboard.

Unfortunately, there is relatively little information about it on the net, there was also talk about an A5 pin, I can't find it either. I ask for help.

regards

[image]

Chopper266, 2022-10-06:
Yest thats correct, you just only need 2pin  only A5(signal) and Grd..

Other hand iam already applyed this wiring but.  When I execd my probe when this knive touch i got alarm hard limit, and this condition make me very confused.. until now iam try to figure out this problem..
I hope any body can give me advice i should to do? Thank you