makerbase-mks/MKS-DLC #24: [HELP] The result is distorded

Nalofox, 2022-02-05:
Hi Everybody,

I'm Nathan, an automatician and I'm here to get help on a problem with a custom laser engraver.

I  built a laser engraver to make customized 3 by 3 round stainless steel water bottles (with 3 lasers in line).
When I want to engrave a water bottle, it starts cleanly, but after a while the line shifts (with and without filling).
I have tried many options, but nothing works as requested.

Here is a test on PVC (it's the same result on a bottle).

- Wanted :

[image]
[image]

- Reality : 
  - Laser 3

[image]

  - Laser 2

[image]

  - Laser 1

[image]

Note that all 3 lasers receives the same TTL signal and if we engrave the same design with same configuration, the result is different on each occurence.

Here is the engraver configuration :
- 3 lasers syncronised and piloted by an TTL signal
- 1 nema 23 for the rotation with belt drive
- 1 nema 23 for the lasers moves
- CNC controller is a makerbase mks dlc v2.0 with 2 external nema 23 driver (MKS TMC2160)

Picture : [https://imgur.com/a/XwitpV4](https://imgur.com/a/XwitpV4)
Config : [https://imgur.com/a/5fvQwcd](https://imgur.com/a/5fvQwcd)

My question is why at a given moment there is a lag...

Thanks for your help.

Nathan

cgmllr100, 2022-02-08:
Is there a chance your pulley on the stepper motor is loose? the problem looks mechanical in nature and a loose pulley, shaft lock collar, or coupler can cause shifts like what you have in your pictures.

Nalofox, 2022-02-11:
Hi,

I don't see lose step, i will add rubber on rotation bar to prevent very small shift.